Armandpl/furuta #24: control the robot using an arduino, control the arduino over serial

Armandpl, 2022-11-24:


Armandpl, 2022-12-11:
Here is what we need in our arduino:
- 5V logic level as this is required for our encoders + our current sensor
- One PWM or DAC pin to control the motor
- One ADC pin to read the current sensor
- Fast serial communication (ideally we should be able to send commands/read sensor value at >>100Hz)
- At least one interrupt pin per encoder, ideally two
- a robust usb connector, a micro usb one is going to break too quickly... 

I've been following [this tutorial](https://docs.m2stud.io/ee/arduino/4-Serial-Communication/) to make [a first version of a protocol](https://github.com/Armandpl/furuta/commit/7c1e3d76d9af1dceb1750ac072aca2fe7186b5e2) to control the robot over serial. Unfortunately, it seems like the serial communication is too slow?

Here is what I'm doing to benchmark the speed:
```Python
@profile
def send_receive(arduino, n):
    bytes_ = [0]*n
    format_string = "<" + "B"*n
    packet = struct.pack(format_string, *bytes_)
    arduino.write(packet)

    packet = arduino.read(n)

if __name__ == "__main__":
    arduino = serial.Serial("/dev/ttyACM0", baudrate=115200)
    time.sleep(2)

    dt = 0
    nb_iterations = 500
    start_time = time.time()

    for _ in trange(nb_iterations):
        send_receive(arduino, 8)
        time.sleep(dt)
```
And the arduino code:
```C
void loop() {
  // wait until we get one dummy packet (8 bytes)
  if (Serial.available() > 7) {
    // read dummy packet
    for (int i = 0; i < 8; i++) {
      Serial.read();
    }

    // send back dummy packet
    for (int i = 0; i < 8; i++) {
      Serial.write(0);
    }
  }
}
```
Here we're only passing around 0 bytes, in a real scenario we would be sending a motor command and getting readings for two encoders and one current sensor.

[image]

The circled number is the one we care about, it is measuring how long it takes to send a command to the arduino and get back an answer. I've only ran this on my laptop so far (thinkpad t470, ubuntu 20).

jefflaichunyin, 2022-12-22:
Arduino Nano with Type C connector: https://www.aliexpress.com/item/1005003880404132.html?spm=a2g0o.productlist.main.1.2cd62c6cV20KUZ&algo_pvid=82351804-c349-41b3-a090-496b2d2ca626&aem_p4p_detail=2022122208101013948492591279900018052921&algo_exp_id=82351804-c349-41b3-a090-496b2d2ca626-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000027378190949%22%7D&pdp_npi=2%40dis%21HKD%2124.16%2119.33%21%21%21%21%21%40211be17616717254108756535d0714%2112000027378190949%21sea&curPageLogUid=EI9anOMGlKBq&ad_pvid=2022122208101013948492591279900018052921_1&ad_pvid=2022122208101013948492591279900018052921_1

jefflaichunyin, 2022-12-22:
Not sure if it's helpful, but you can pick suitable baudrate from [this table](https://trolsoft.ru/en/uart-calc).

jefflaichunyin, 2022-12-22:
Maybe you can test [Serial.readBytes()](https://arduinogetstarted.com/reference/serial-readbytes) as well.

Armandpl, 2022-12-22:
@jefflaichunyin this is much appreciated, I'm going to look into that!

closed 2023-12-02 by Armandpl