Armandpl/furuta #54: add and evaluate current sensor

Armandpl, 2024-01-17:
- [ ] simulate current sensor in simulation
- [ ] train with/without current info **in sim** and see if it provide an advantage
- [ ] it the following can
- [ ] update firmware and com protocol to work with the current sensor on the real robot
  -  could be that the current sensor doesn't help in sim but would help in real life so this can be done in parallel
- [ ] train with/without current sensor **in real life**
- [ ] decide if we want to keep the current sensor or discard it
  - if we discard it, revert changes to the code but keep track of experiments somewhere

pierfabre, 2024-01-21:
Our current sensor outputs 5V analog which is too much for the seeduino. We then need to go through a ADC and then through a logic level shifter. This makes us want to maybe use a current sensor that outputs 3V max 
[CS0350BL](https://www.cuidevices.com/product/resource/csxx50bl.pdf) would be a good candidate.
[image]
On this figure, If is 3A which corresponds to our motor stall current.
It cost around 25€

Armandpl, 2024-01-21:
Additional issue: reading the ADC is slow (~300Hz)

Maye it can be set to continuous mode? What does that mean exactly? see https://github.com/RobTillaart/ADS1X15/blob/master/examples/ADS_continuous_4_channel/ADS_continuous_4_channel.ino

pierfabre, 2024-01-22:
Might have found something!
https://cdn-learn.adafruit.com/downloads/pdf/adafruit-ina219-current-sensor-breakout.pdf 
range : +- 3.2 A
logic level : 3V accepted
has a ADC inside

pierfabre, 2025-07-15:
We tested it and it slowed down the control loop. I used it for system id but we don't need it for the control part for now. closing

closed 2025-07-15 by pierfabre