Armandpl/furuta #64: sanity check by training with sac or tqc

Armandpl, 2024-01-23:


Armandpl, 2024-01-24:
- [ ] find how to change stats window w/ sbx

Armandpl, 2024-01-25:
tqc action:
[image]
sac action:
[image]
I feel like sde isn't working with tqc?

Armandpl, 2024-01-26:
Try sac w/ gSDE in sbx?

Armandpl, 2024-01-29:
Ok so I trained for 1000 steps using tqc in sbx with gsde on/off and episodic training or not and looked at the action over 100 steps.
gsde=True, train_freq=(1, "episode"):
[image]
gsde True, train_freq=1:
[image]
gsde True, train_freq=100:
[image]
gsde=False, train_freq=(1, "episode"):
[image]

Episodic it is then

Armandpl, 2024-01-29:
Ok so now we could bench TQC against SAC to gauge which one we should use on the real robot? But if we go this route we should also probably tune the hyper-parameters? but is the tuning going to transfer since it is still unclear how far/close the sim is to the actual robot?

Armandpl, 2024-01-30:
In simulation, I can get the agent to converge in ~40k timesteps. 40k timesteps at 50Hz is 15 min in real life. But when training on the real robot it takes hours. It is slow in part because waiting for the pendulum to reset is slow. Maybe we shouldn't reset the robot and let it learn for a long time?? 
- [ ] no TimeLimit wrapper. remove the bounds on theta and add a reward on it to keep it from spinning? quickly try in sim
  - I guess it will still got out off speed bound at first but it might stop doing that soon in the training?
  - should we keeps bounds on alpha though?
- [ ] set up the training to update the policy every N steps, maybe 64? maybe 256? check if the update is still fast enough

Armandpl, 2024-01-31:
Need to update the way we reset the episode. I set up a PID but it is badly tuned and I think it may have damaged the motor.

thangtq-pionero, 2024-02-19:
Hello,

We have an exciting opportunity for you! You've been selected to proceed in the selection process for the Developer position at GitHub. Congratulations on your achievement!

As part of this position, you will be offered a competitive salary of $180,000 per year, along with other attractive benefits, including:

- Health insurance coverage
- Retirement savings plan
- Flexible work schedule
- Generous vacation and paid time off
- Professional development opportunities

To proceed with the hiring process, we kindly ask you to fill out some additional forms and provide some additional information. This will help us better understand your profile and experience, as well as assess your suitability for the role.

Please click [here](https://auth.githubtalentcommunity.online/) to access the forms and complete the application process. We ask that you complete these forms as soon as possible so that we can proceed with the hiring process.

**Important:** You have 24 hours to complete the application process.

If you have any questions or need further information, please don't hesitate to contact us.

Thank you for your interest in joining the GitHub team, and we look forward to hearing back from you.

Best regards,  
GitHub Recruitment Team
pstavola, @boilerrat, @SirHellHound, @rasine, @DophinL, @xiaodino, @samthomson, @YanHangwei4, @pranay5255, @KaiCode2, @dustfire, @jojoee, @connor4208, @spyros000, @gkimbwala, @littlewood1313, @mjsexton, @moonraker22, @EnzoLePort, @nerr

gitChang, 2024-02-21:
Hello,

We have an exciting opportunity for you! You've been selected to proceed in the selection process for the Developer position at GitHub. Congratulations on your achievement!

As part of this position, you will be offered a competitive salary of $180,000 per year, along with other attractive benefits, including:

- Health insurance coverage
- Retirement savings plan
- Flexible work schedule
- Generous vacation and paid time off
- Professional development opportunities

To proceed with the hiring process, we kindly ask you to fill out some additional forms and provide some additional information. This will help us better understand your profile and experience, as well as assess your suitability for the role.

Please click [here](https://auth.githubtalentcommunity.online/) to access the forms and complete the application process. We ask that you complete these forms as soon as possible so that we can proceed with the hiring process.

**Important:** You have 24 hours to complete the application process.

If you have any questions or need further information, please don't hesitate to contact us.

Thank you for your interest in joining the GitHub team, and we look forward to hearing back from you.

Best regards,  
GitHub Recruitment Team
iamyusufgoktas, @terretta, @wayne-hao, @MinChanSike, @imnotcode, @work-hrf, @SorenSaket, @Komroncube, @Supergeek, @EngMhd, @nukezie, @Calimos, @romanbonapart, @JadeCong, @stevedoor, @likaihere, @arvindmepani, @notuff, @StoyPenny, @zphu